fpv-wtf/wtfos #6: Feature Request: Enable Audience mode regardless of 50Mbit/s mode

PhilippMolitor, 2022-07-01:
One of the most dreaded issues when flying in a group is being unable to join in with the audience mode when the goggle is still set to 50mbit/s mode. Connecting a quad usually is not a good idea as it will probably cost a pilot his signal.

I think it would probably be an easy patch to enable 25/50 mbit/s switching regardless of being connected to a vista/air unit. This way the patch could also just always allow audience mode and switch to 25mbit/s when selecting that.

funneld, 2022-07-05:
The algorithm to fix this.

`setprop dji.glasses_service 0

write 0x00 to address 0xc0 in /blackbox/glass_user_setting.dat

setprop dji.glasses_service 1`

and some logic when.

funneld, 2022-07-07:
If you want to test an early experimental version. Not sure it'll work on V2s. Hold down the "channel down" button for 8-10s to switch.
https://github.com/funneld/djifpv_offline_audience_switcher/releases/tag/v0.0.2