fpv-wtf/wtfos #7: Communication error on v1 goggles to air unit lite

burzl89, 2022-07-01:
when i want turn on auto temp control or other device setting nothing happend.
also settings for mbit focus ect.

camera settings also not able to change.

i booted without hack. all ok.
then i turned live audio service in goggles off. reboot- all ok. i have this on becouse i have a air unit also.

hope this helps

my setup:
goggles v1 fcc hack
air unit lite fcc hack nebula pro solo, without fc

funneld, 2022-07-02:
Are you sure your AU wasn't just overheated?

AU lite doesn't have a microphone so the live audio won't work. It'll only work with the standard big airuinit.

I can confirm that I don't have any problems with the auto temp control. Only when set to 1000 or 1200mW and playing with the ON-OFF auto-temp toggle it takes about 30s to enter low power when set from OFF to ON. At 700mW or lower it does it instantly.

burzl89, 2022-07-02:
Hello!

Yes I am sure. I cool aktiv. Temp sensor was about 46 degree C.

I know that no microphone is present. But disabling live audio in the goggles was solve the problem. I am not sure why. I think data frame addresses ore so are others as in the air unit and collide with audio stream packets in Vista models. I have No plan :) 

Problem was not heat, it was not possible to change settings. Like temp control, transmission Bitrate, power ect

funneld, 2022-07-02:
I'm thinkig it's possible it kills the dji_hdvt_uav or dji_shuttle on the Vista because there's not a microphone device when and it tries to open it.

ramiss, 2022-10-16:
I've been experiencing the exact same issues on my caddx units,  I'll try disabling the audio package. Thx