fpv-wtf/wtfos #17: Failed health checks

langhua1duoduo, 2022-09-18:
Failed health checks

Don't worry, this is normal and can happen due to various reasons, including bugs in the underlaying operating system. The configurator can help you fix these issues. When attempting an automated fix, your device might reboot in the process. In some specific cases you may need to re-root your device after a reboot, before health checks can continue.

+ set -e
+ check
+ grep /dev/loop
+ mount
+ grep -q '#wtfos' /system/bin/prestart_dji_system.sh
+ grep -q '#these are leftovers' /system/bin/prestart_dji_system.sh
+ echo 'wtfos is installed but system.img is not mounted, please try re-installing wtfos'
wtfos is installed but system.img is not mounted, please try re-installing wtfos
+ return 1

j005u, 2022-09-18:
@langhua1duoduo are you on our Discord?

If so, I'm @Joonas. Get in touch, we need to look at what went wrong a little closer.

langhua1duoduo, 2022-09-18:
@j005u 
The configuration program prompts me to reinstall WTFOS, but "prestart_dji_system. sh" contains wtfos related content and cannot be install. I want to uninstall WTFOS and reinstall it, but WTFOS does not run successfully, so the configuration program does not allow the uninstallation.

I checked the content of "prestart_dji_system. sh", but I didn't quite understand it. I tried to load/dev/block/xxxx/blackbox into/blackbox/test, and found no WTFOS folder. I think the DJI system restored it to the factory state.

Then I copied the Blackbox folder from my friend's glasses (his glasses are also root and installed WTFOS) to my glasses, and then it worked normally.

langhua1duoduo, 2022-09-18:
When this fault occurs, there is no "waiting osd..." on the glasses screen, and all settings (language, low voltage alarm, custom OSD, etc.) are restored to the factory state.

j005u, 2022-10-02:
This health check was disabled because it was causing other issues. Please open a new one if you're still having problems.

closed 2022-10-02 by j005u