fpv-wtf/wtfos #18: Video out broken

Landairfpv, 2022-09-19:
Goggles v1 video out via digiview app doesn't work after flashing wtfos and osd package. Is there a work around to fix this?

j005u, 2022-09-19:
No workarounds right now.

Needs some research into what actually breaks. We've been busy with bigger issues, but should be able to look into this soon.

mcaraballo, 2022-09-30:
Was this officially fixed after the official release? I thought I heard yes from somewhere on Youtube but can’t recall. These gents are working so hard for free I want to refrain from being a pest

mcaraballo, 2022-09-30:
Disregard.  It works and I’m using widefpv.com (Amazing tool)! I had trouble with one of my Mac’s and presumed it was a wtfos issue but that doesn’t seem to be the case

bri3d, 2022-09-30:
This tracks - we did some research over the last few days and it seems like most of the reported issues post-wtfos with video out relate to USB driver conflicts / device enumeration issues. Happy flying, glad it’s working for you!

j005u, 2022-09-30:
@bri3d adb interface remains number 3, no issues with enumeration.

It appears to be purely drivers. The one thing where I once had glasses crash with video out seems to be a thing of the past.

closed 2022-09-30 by j005u

pierluigizagaria, 2023-08-07:
I cannot get any video out after wtfos, can this be the cause?

DangerD1024, 2024-12-20:
Any update on this?