fpv-wtf/wtfos #19: OSD Layout messed up if goggles are rebooted

tonyyng, 2022-09-23:
After the initial connection, where the OSD elements are in their correct location, I disconnected my goggles battery to test the behavior with a battery swap.

After the connection between the goggles and the air unit is re-established, the OSD is unusable.  There are some random digits scattered around the screen.  If turn the power to the air unit OFF/ON, the OSD starts working again.  I have the air unit on a switch, so I'm not rebooting the FC.  

I'm using WTFOS with iNav 4.1

Thanks

tonyyng, 2022-09-23:
[image]
Using a channel to turn the OSD off and back on fixes the OSD display.

closed 2022-09-23 by tonyyng

mavenius, 2022-09-23:
Awesome workaround; thanks!