fpv-wtf/wtfos #24: can't find the caddx vista(dji air unit lite) when root device

Luukk1919, 2022-09-27:
I had root my DJI goggles V1,
when I want to root my caddx vista it is not shown on the pop list box.
but my computer can recognize that and it's shown in the DJI assistant app also.
my computer is 2019 macbook pro and it's still not shown when I try to root on a win 10 labtop.
P.S.  I use google  chrome app and all are the least version.

I had spent two nights to try to solve it and can't figure out how to do.
I installed the CDC serial driver and the bulk interface driver on my computer and it doesn't work.
please help me. 
thanks so much.

Luke

EVilm1, 2022-11-03:
Hi, You can try to reboot, do not open DJI Assistant to avoid occupying the serial port.
If this does not work, make sure you have a usb C cable with data transfer.

unrealkingdom, 2023-02-09:
I have same problem..with other USB i was try,but with other vista lite it'oke..air unit lite firmware 0608 with runcam wasp,
this is message i found "Could not connect to device, make sure that adb server is not running on your machine (adb kill-server) and that you are not connected in another tab/window." anybody can help me?

j005u, 2023-06-13:
The configurator has since been significantly improved and you should no longer run into these issues.

If you do then please open a new issue including your trace-id from the bottom left of the page.

closed 2023-06-13 by j005u