fpv-wtf/wtfos #25: after installing permanent fcc 1200mw app no more 8channels only 1200mw

artemen, 2022-09-29:
after installing permanent fcc 1200mw app no more 8channels only 1200mw
installed on both googles v2 and the air unit

j005u, 2022-09-29:
@D3VL-Jack ping.

D3VL-Jack, 2022-10-02:
Hi @artemen,

I'm unable to replicate this issue, could it be that your transmission was set to 50mbps instead of 25mbps?

Cheers!

kbzowski, 2023-07-30:
Can also confirm.
DJI google v1 + air unit
fcc-unlock 1.0.6
Firmware 01.00..0608
1200W/50mbps is available, but only channels: 1,2,3,8

1. Removed fcc-unlock on both and put naco/naco_pwr.txt - nothing changed.
2. Setting google to 25mbps unblocks all channels

Roman-Golod, 2023-07-31:
> Can also confirm.
> DJI google v1 + air unit
> fcc-unlock 1.0.6
> Firmware 01.00.0608
> 1200mW on 50mbps is available, but only channels: 1,2,3,8
> 
> 1. Removed fcc-unlock on both and put naco/naco_pwr.txt - nothing changed.
> 2. Setting google to 25mbps unblocks all channels 

As I know, it's normal, because 50mbps uses 2 channels in parallel, so their quantity is decreasing.

kbzowski, 2023-07-31:
> As I know, it's normal, because 50mbps uses 2 channels in parallel, so their quantity is decreasing.

Oh, that was not obvious :) So it works fine. I was confused by the first post in this thread.