fpv-wtf/wtfos #27: Air unit stuck in a connected/disconnected loop (no video)

Spit29, 2022-10-02:
I can't seem to get my dji air unit to work after rooting and installing WTFOS.. I don't get video and it seems to just boot loop. I am getting a repeating on and off usb connection. I'm not able to connect to the air unit via usb though.. Any help is greatly appreciated.
I was able to root it and install the OSD mod but now it just loops...

LouDnl, 2022-10-02:
How are you powering the AU?

j005u, 2022-10-02:
Try booting while holding down the bind button. It should start in safe mode and allow you to re-connect in assistant.

Spit29, 2022-10-02:
> 

Powered by a new speedybee F7 v3 + 6s battery. It worked before the Wtfos. Holding the bind button while powering doesn't seem to do anything. Led us solid red

Spit29, 2022-10-02:
Got it! Tried it on another drone and it worked. Turned out the wiring is not correct when using the plugs the Speedybee f7 v3 came with.. Very annoying..

closed 2022-10-02 by Spit29