fpv-wtf/wtfos #30: Googles V1 and Air unit don't show the OSD after installing WTFOS stuff, dont show quad battery status and DJI assistant 2 show error: Cannot load firmware list (4-10)

ObiVanDunobi, 2022-10-14:
Hi,
today I tried to unlock the full betaflight OSD through WTFOS.
Root process, also installing WTFOS and packages (FCC and msp_osd) done, CLI and other Betaflight was set according to JB video.
But alfter all of that, OSD not showing up, quad battery voltage is gone and when I trying to open firmware list at DJI Assistant 2 it shows error with code:
Cannot load firmware list. (4-10)
This error show for both devices - googles and air unit.
In googles I just see in the right bootom corner: OSD WAITING...
What can I do for showing battery voltage at least please?

ObiVanDunobi, 2022-10-15:
SOLVED
Dumb user  :/
In CLI comands I didnt write SAVE so the CLI never saved up MSP settings.
But DJI assistant still didnt wok. I can live with that if I will never need DJI assistant 2 any more. Hope I will never need it again :D
Sorry for false issue.
Thak you
Juraj

closed 2022-10-15 by ObiVanDunobi

magizusebas, 2022-11-05:
Hello, i am having the exact same problem, but after saving the cli for MSP, i still cant see the betaflight osd, i dont know if you could help me or if you were able to find a way to connect to dji assistant 2

ObiVanDunobi, 2022-11-05:
> Hello, i am having the exact same problem, but after saving the cli for MSP, i still cant see the betaflight osd, i dont know if you could help me or if you were able to find a way to connect to dji assistant 2

Hi, make sure that you have all betaflight settings according to JB or Oscar Liang and your MSP port should be -1 from your actual UART where you have air unit soldered.
My OSD works well now. 
I think you have to turn off Custom OSD in DJI Goggles.
For dji assistant, I dont use it any more.
Still not able to change firmware because of error message (Cannot load firmware list).
But there is a way how to able it again.
If you have installed WTFOS, you have to run one phrase at CLI.
See this from 5:30.
https://www.youtube.com/watch?v=vubarw7BJkw
Good luck and let me know 😉

Chupokabra5, 2023-11-04:
The same thing doesn’t work at all I’ve tried the cli and the preset and 4.4

Chupokabra5, 2023-11-04:
Betaflight 4.4*

ObiVanDunobi, 2023-11-04:
Hi,

I dont remember how I made it but its work for me now.
I know I made a one mistake:
I didnt hit the Save button after putting commands in CLI.
So try to do all the steps again and dont forget save CLI commands.
Let me know if it helps.