fpv-wtf/wtfos #33: How to enable live-audio on DJI FPV Goggles V1

ez4fm, 2022-10-31:
I have installed this package, but I don't know how to enable it

j005u, 2022-10-31:
It should just work, if you have a full size air unit on the other end.

It can't work with Vistas, as they don't have a microphone.

ez4fm, 2022-10-31:
Thank you.
I'm using RunCam Link Wasp now, I'll try it on DJI air unit again.

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

gmoska, 2022-11-08:
Could someone share the pinage of the audio? I tried live-audio without sucess :(

j005u, 2022-11-08:
Pinage? You just need to plug in headphones, install the package on your goggles V1 and use a full size Air Unit.

@gmoska if you have a problem, please open another issue with a detailed description of what you did, what devices were involved and what happened.

Closing this issue as the original problem seems to have been using a vista.

closed 2022-11-08 by j005u