fpv-wtf/wtfos #34: Updated to 01.02.0015 not able to root

joe-0310, 2022-11-01:
Hello,

I am not able to install wtfos because I updated to 01.02.0015, are you working on a hack to root from this version or is this to be considered hopeless?

ako82, 2022-11-08:
Stay tuned. There is something in the cooking.
Don't upgrade any further.

...or some similar reply would be more than enough to this question. I would also like to know if there is some hope in the horizon :(

j005u, 2022-11-08:
Feel free to upgrade past 0015, it doesn't make a difference.

We've stated in several places that we are in fact working on a solution, but it's going to take some undetermined amount of time to get to anything publicly releasable.

No updates means we're still working on it.

closed 2022-11-08 by j005u

ako82, 2022-11-08:
Ahhh... that's something! Thanks for all your efforts, take all the time you need!