fpv-wtf/wtfos #35: GPS Home Direction Arrow not working

Anton33455, 2022-11-02:
Instead of GPS home direction arrow, I only get a dash on the screen.
Fonts are not possible to change as well.

Any idea if I am missing something?  Both goggles and vista unit are flashed with latest WTFOS
Rest of OSD works, but the font is really ugly.

EVilm1, 2022-11-03:
Hi, you can check this video around 11:30 : https://www.youtube.com/watch?v=hNOA0kUjKhY
this will answer your question ;)

Anton33455, 2022-11-03:
So no fonts as of now.. Funny that I like MaddsTech font.  Mine looks like
retro.  How did he get that font?

j005u, 2022-11-03:
Fonts are currently handled by placing the correct file on your SD card. BF font selection is not honored.

There's a few fonts linked in the msp-osd README and instructions here: https://github.com/fpv-wtf/msp-osd#suggested-third-party-fonts
There's also SneakFPVs font at: https://drive.google.com/drive/folders/1buxrXqhU46AxE3fwaFDsMb97IiGLVa95

A future update will bring font installation into the configurator.

closed 2022-11-03 by j005u

cmanley, 2023-01-31:
Are there previews of the fonts somewhere?