fpv-wtf/wtfos #36: Feature Request: support for live-audio on DJI FPV Goggles V2

sleepycat233, 2022-11-08:


CoefficientofSquish, 2023-01-22:
I found a package for live audio enable on v2, but it was pretty difficult to install (for me anyway)and I get the feeing that maybe it was intentional to filter people out so they wouldn’t damage their goggles.  I did manage to install both successfully but then I got stuck on the FPV side and I’m hoping that was unrelated because my goggles lost and regained power a bunch of times in short succession.  I’m about 60% certain I could hear audio in the headphones before that happened but I had them powered from a wall outlet so I couldn’t get too far away from them.  When I was attempting to adjust volume in CLI I got error messages from the second method that was suggested but the first method seems to have worked.  The second one may have been for the other device.  If I can get it working I’ll come back and share what I find but so far I haven’t seen any info anywhere else so if this something everyone already knows about and there’s a faster way please clue me in.

sobelek, 2023-03-03:
@CoefficientofSquish What package did you install?

D3VL-Jack, 2023-03-03:
@sleepycat233 & @sobelek, You can find Live Audio for V2's here 👉 https://github.com/funneld/djifpv_live_audio_V2
@CoefficientofSquish - Getting stuck in FPV mode is absolutely due to the rapid power cycling, any issues when using this specific package are best asked over there 😄