fpv-wtf/wtfos #37: Switching recording mode from caddx vista to air unit

Ciurrone, 2022-11-13:
I fly with fixed wings and quadcopters, on fixed wings I use air units and on quadcopters I use caddx vista.
I like to record videos on the air units, but when I change from quadcopter to fixed wing, the recording on the air unit is disabled and I often forget to reactivate it from the glasses.
there would be no way that the glasses automatically switch the recording mode according to the unit used?

LouDnl, 2022-11-15:
It's on my todo list, just haven't had the time yet.
https://github.com/LouDnl/wtfos-auto-record

Ciurrone, 2022-11-15:
> It's on my todo list, just haven't had the time yet. https://github.com/LouDnl/wtfos-auto-record

I hope you will succeed soon