fpv-wtf/wtfos #38: Feature Request for av in screen resolution

kakashi2018, 2022-11-17:
Hi,
 I just wanted to know if wtfos is capable of changing the aspect ratio or the resolution when using av in v1 or v2 goggles? it is so inconvenient for many users who are using analog with v1 and v2 goggles as the image size is much smaller

willertan, 2022-11-26:
LOL I was searching "wtfos Goggle V2 av-in enlarge" and lead me to this post. I fly both digital for 5 inch & analog for small quad, rc boat, rc car/truck (because it is cheaper). YES i needs this feature too.

j005u, 2022-11-26:
I'm afraid this one is rather difficult. While it might be technically possible to figure out where the RTOS composites the av-in feed,  it's buried somewhere deep in the binary and may be difficult to modify. Especially when it comes to scaling the image to fill all screen pixels.