fpv-wtf/wtfos #39: Is it possible to fcc hack the DJI FPV drone on latest firmware?

deethee, 2022-11-27:
I love the DJI FPV drone due to the gimbal, superior image, and position hold for flying in mountains. Since I was forced to upgrade to latest 1.04 firmware to support DJI o3 Air unit I traded in the b3yond fcc hack that is now nolonger working. Is it technically possible to do a fcc hack by wtf for the DJI FPV drone? Is it possible to roll back the DJI FPV firmware only and not the others with the root?

eabrega, 2022-11-29:
Greate! I have this question too.

RealTadango, 2022-12-02:
As i understood there is no way to root / setup wtf once you have fw .15 or newer.

deethee, 2022-12-02:
I have root than upgraded

RealTadango, 2022-12-02:
> I have root than upgraded

Yes but the issue is that many have already the upgrade without root. There is no solution for that. My goggles were replaced by support with a version that had the newer firmware :(

j005u, 2022-12-02:
No, not without a downgrade method. We are working on it.

closed 2022-12-02 by j005u