fpv-wtf/wtfos #40: Have voltage on OSD but no RSSI, LQ or cronometer

Helix-in-Wonderland, 2022-11-28:
Hello good people

I've just installed on DJI V1 goggles, WTF, the OSD and power packages. From the items I chose on OSD, only disarmed and voltage are displayed, no LQ, RSSI dbm, warnings or cronometer. I'm using ELRS Happymodel EP1 and sensors are working. Any ideas? Thank you.

kakashi2018, 2022-11-28:
@Helix-in-Wonderland Did you do the msp and serial port setup in betaflight?

Helix-in-Wonderland, 2022-11-28:
Yes I did. And voltage is showing. Thanks

Helix-in-Wonderland, 2022-12-16:
Well, can't get no one to answer here or any other place. I'll go back to the craft name hack. No wtf for me.

hickscorp, 2023-01-17:
It'd be useful to get your BF configuration @Helix-in-Wonderland . Maybe show us what `diff all` shows?
It might be that you have `feature -TELEMETRY` or some of the `telemetry_disabled` stuff.

Helix-in-Wonderland, 2023-01-18:
Thank you. I'll try to do that as soon as possible as I'm not with access to the quad at this moment. Cheers