fpv-wtf/wtfos #45: Add support for Goggles 2

bakwc, 2022-12-05:
Add support for [Goggles 2](https://www.dji.com/goggles-2) (or "avata" goggles, not V2)

j005u, 2022-12-08:
This will require an all new exploit for the Goggles 2 to root.

We are working on it, but no ETAs whatsoever.

bakwc, 2022-12-08:
And what about Walksnail? Do you know anything about it? Is there a way to connect to it and make some custom overlay?

j005u, 2022-12-08:
The earliest FWs had a root password of "artosyn" and were indeed modifiable. Walksnail removed that access and nobody on the team has been motivated to spend time playing cat and mouse with them. Their blatant GPL violations don't help with the latter.

wendal2841, 2023-03-31:
do you plan to support goggles 2 ?

j005u, 2023-03-31:
> do you plan to support goggles 2 ?

Maybe, if we find a new exploit for the Goggles 2.

No timelines or promises.

FPVZaphod, 2023-04-12:
Would be outstanding. It's a shame, DJI should hire and pay you guys for fixing their products...

MR444, 2024-03-24:
One year later... are there any news about supporting O3 air units or Goggles 2? Or has this project ended?

The situation with DJI goggles is really unsatisfying at the moment. With the goggles 2 or Integra, you do not have WTF.OS and not AV-in, with the Goggles V1 there is no support for O3, with Goggles V2 you cannot use with O4 systems and cannot record the OSD data of the O3...

I hope another system will reach DJI's quality level soon.