fpv-wtf/wtfos #46: Feature Request: DJI 03 Air Unit support for V1 Goggles

mattcattell, 2022-12-12:
Is it possible add support for the DJI 03 Air Unit into the older V1 DJI Goggles as part of this project?

j005u, 2022-12-12:
Unfourtunitely that's a firm no.

While the hardware is theoretically capable at 5.8 gig, we simply don't have enough tooling (or exploits for running unsigned code from flash directly) to hope to achieve this.

mattcattell, 2022-12-12:
Thats a real shame. 

So, is the problem that you can't execute a large amount of new code flashed onto the goggles with the current exploit? can you only modify or add to existing DJI functions.

j005u, 2022-12-12:
We can only modify the RF RTOS at runtime, not in flash.

This would require somehow backporting the new O3 RTOS to the V1 Goggles. From binary form, to binary form, without any sources or tooling involved.

While not beyond the laws of physics, it would be a multi year undertaking at best.

mattcattell, 2022-12-12:
I see, I understand the problem. 

Is there any difference between the v1 & v2 goggles regarding micro & toolchain, just thinking could anything be taken from the latest binary/firmware for the v2 goggles (not goggles 2) ?

j005u, 2022-12-12:
Different signing keys and somewhat different hardware. Wouldn't work out of the box even if the keys were the same.

It's truly a dead end.