fpv-wtf/wtfos #47: Goggles V2 locked waiting for an update

youprint, 2022-12-21:
I bought an DJI FPV combo, but i did the mistake to connect it to DJI Fly app.
So now it's locked, impossible to go into the menu. the actual firmware is  V01.02.0000
i tried drone hack to flash V01.02.0001 or V01.02.0020 but it's not working.
it's a trap from DJI to force update, some user after many attempts got it work with drone-hacks but for me it's not working.

Do i need to let DJI win and flash V01.02.0015 or is there a hope in the future ?

RealTadango, 2022-12-21:
See https://github.com/fpv-wtf/butter :)

youprint, 2022-12-21:
unfortunatly :  Waiting for DJI Recovery device to enumerate.. tryed all usb ports and different cables

RealTadango, 2022-12-21:
You closed the DJI assistent right? Drivers installed?

youprint, 2022-12-21:
of course

j005u, 2022-12-21:
> unfortunatly : Waiting for DJI Recovery device to enumerate.. tryed all usb ports and different cables

You should try to find another PC to try on. So far we haven't been able to identify anything wrong with the driver installer (presuming you ran it), the issue seems to be that the DJI recovery USB interface behaves in a slightly non standard way and some USB controllers(/their drivers) simply can't handle it.

youprint, 2022-12-21:
i don't think it's a usb pb, i tried on 3 differents windows computers and same
i flashed V01.00.0606 using this link : https://mega.nz/file/Uz4V1L4Q#XQAXasHy95XYuhj1Mc4yac5Gg-uX2kmrmay_yr92_iI found in butter readme : it flashed succefully but buttons are not recognized

So i tried V01.04.0000 still found in butter readme and now the buttons work but i'm unable to go into the menus.

j005u, 2022-12-21:
Right, okay, that's a different problem.

Update via assistant to the latest version, get everything unlocked and then downgrade back using butter to root.

For butter closing as soon as you plug in the device you can try running the .bat file from the windows command prompt. It might show some helpful errors.

youprint, 2022-12-21:
ok

youprint, 2022-12-21:
impossible to upgrade via assistant i have an error after dowloading the firmware. DJI wants us to use DJI fly app.

j005u, 2022-12-21:
Then use the Fly app.

You can downgrade after now that butter has been released.

youprint, 2022-12-21:
my new pb with fly app is that they required to connect the drone and the controller and when i do that they try to force an update to the drone. i can't upgrade the drone or i m gonna loose my nfz hack
i'm f.......

youprint, 2022-12-21:
finally i managed to install V01.05.0000 via the fly app without updating the drone. 
i downgraded via butter to V01.00.0606 and switched to DIY
i rooted and installed WTFOS.
thanks so much for the help, i need to donate now.

closed 2022-12-22 by j005u