fpv-wtf/wtfos #53: Cannot update through Assistant my rooted units

Kamdozer, 2023-01-12:
Today i tried to update my rooted units and vistas to make them compatible with Goggles 2. Unfortunately with no success. Both type unit and vista were recognized by DJ assistant (both consumer an FPV versions), but no firmware was listed. Only error to obtain firmware 4-10 error. I tried a lot of things. In the end I decided to update an unrooted vista and voila - no issues. I am on mac. So obviously DJI has locked in some way updates for rooted devices. I guess my only option is to remove the root, but could not find info how to do it on the units. My goggles v.2 will remain rooted for other reasons, but I move to goggles 2 and now i can not update my rooted hardware for the purpuse. Any suggestions guys?
[image]

xviews, 2023-01-12:
1. remove WTFOS
2. CLI wtfos-remove-adb

Kamdozer, 2023-01-14:
Thanks a bunch. It worked!

closed 2023-01-14 by Kamdozer

tiac0o, 2024-11-19:
> 1. remove WTFOS
> 2. CLI wtfos-remove-adb

I did the step 1 and success, but when i exec the CLI command, it says can`t find the 'wtfos-remove-adb' command