fpv-wtf/wtfos #54: after install Wtfos, o3 switch menu diasppear

tillanto, 2023-01-13:
After i install wtfos  in my dji gog v2 o3 switch menu disappear and i have to connect to assistant to make it fix but after wtfos no longer in the gog.

whyzermans, 2023-01-16:
Hi. I had similar issue as you are having. Rooted and installed everything "by the community book" but still encountered an issue. If I connected to DJI FPV Series assistant in DJI FPV mode everything would flash OK. I would see all four modes including O3, but as soon as I wanted to install WTFOS the configurator gives me "slot 2 active - wtfos should be in slot 1" - attempt fix" message.

If I picked attempt fix - wtfos would install and MSP OSD in DIY MODE for Vistas would be working perfectly, but no option anymore to switch to O3. And if you try to reflash, same loop would occur - but then I encountered this and it works for me perfectly.

Hope that helps.

**Installing wtfos after upgrading downgrades the firmware**

If you update to V01.04.0000 (or similar) after rooting and then proceed to install wtfos, sometimes you can get kicked back to V01.00.0606 and loose O3 AU support. This happens because sometimes Assistant will only flash the second slot. Use the following commands in the configurator CLI:

unrd slot_1.status_successful 0
unrd slot_1.status_active 0
unrd slot_2.status_active 1
reboot

Then refresh the configurator. Health checks will come up saying slot 1 is corrupt. Press attempt fix, slot 1 will be re-flashed to the correct version.

[](https://github.com/fpv-wtf/wtfos/wiki/Troubleshooting)

j005u, 2023-01-20:
Yup, there's an issue where DJI Assisatant will sometimes only flash one of the two firmware slots, thus you get kicked back to the older FW without O3 support when installing wtfos.

The above fix should work, if not please re-open the ticket.

closed 2023-01-20 by j005u

zvikaf, 2024-02-24:
got the same issue on v2, DJI menu disappeared after installation