fpv-wtf/wtfos #68: Could not connect to device, make sure that adb server is not running on your machine (adb kill-server) and that you are not connected in another tab/window.

levin-wy, 2023-02-20:
When I connet the DJI  Air Unit,it show me this:
Could not connect to device, make sure that adb server is not running on your machine (adb kill-server) and that you are not connected in another tab/window.

vjvenida, 2023-03-09:
I have the same error..  Are you able to fix this error? thanks

levin-wy, 2023-03-09:
> 

I haven't found a way.

vjvenida, 2023-03-09:
I'm trying this fix right now. I think you should too.. 

https://github.com/fpv-wtf/wtfos-configurator/issues/230

vjvenida, 2023-03-09:
dude, fixed mine.. its now working.
Run butter, update dji assistant 0606 then re-root goggle then install wtfos.
https://github.com/fpv-wtf/margerine/wiki/Troubleshooting#wtfos-install-stuck-in-a-loop

vjvenida, 2023-03-09:
WTFOS Team, Thank you so much for this!

yak-54, 2023-03-12:
Hi i have this issue as well with an air unit 
i did update v1 googles is fine i did one air unit as well its fine 
went to do one more air unit and all i get is this 

Could not connect to device, make sure that adb server is not running on your machine (adb kill-server) and that you are not connected in another tab/window

suprchunk, 2023-03-20:
> 

Same here. Worked last week, on the same unit.

j005u, 2023-04-09:
This issue should be fixed by going to https://fpv.wtf/root and re-rooting. Sometimes you may have to do it multiple times during an install process.

Tracking this as a recursion here: https://github.com/fpv-wtf/wtfos-configurator/issues/176

closed 2023-05-16 by j005u