fpv-wtf/wtfos #77: please catch non-activated /  non-spying Vistas.

AndKe, 2023-05-19:
I had several failures/stops after 10 retries on step two.

00:01 Attempting Step 1...
00:04 - Found Device: LT150, Version: 01.00.0606
00:04 Step 1 - Success! Rebooting...
00:21 Attempting Step 2...
(here was a text about 10 retries)

The reason: I did not "activate" the product, so the manufacturer's evil reasons were not satisfied.

Please consider adding the logic to mention this possibility when step 2 fails after10 attempts.

j005u, 2023-05-19:
Thanks, that is indeed a good suggestion. I've filed an issue for this on the configurator repo: https://github.com/fpv-wtf/wtfos-configurator/issues/374

I should note that we stay away from explicitly working past activation because DJIs activation servers check the serial numbers during activation against a blacklist of stolen gear. We don't want to help bad people activate illegitimately obtained hardware.

We've studied the relavant behavior of DJI Assistant and I can assure you they don't send your flight stats, GPS, data, media or anything of the kind to their servers. If you don't want DJI to know how many devices you have (activation does get bound to your account) you can always use a throwaway e-mail address + VPN, but you may run into issues with warranty in that case.

This only applies to the old generation DIY gear. Anything that talks to the DJI Fly app (O3, commercial drones) are indeed a complete privacy nightmare. Especially if you opt in to _any_ of their customer experience improvement programs.

Closing this issue in favor of the configurator one.

closed 2023-05-19 by j005u

AndKe, 2023-05-19:
@j005u  Thank you.
I was not concerned about sending data, as that requires an app/internet connection, and for FPV use - I simply don't use any DJI apps.
I am more concerned about stuff like DJI AEROSCOPE and similar use signals of any air units, even lost/stolen ones that are activated, revealing the original owner.  - and possibly make the original owner a target for unwanted attention.
I don't see it as very unlikely that other internet connected DJI drones sniff and report other nearby traffic as well.

j005u, 2023-05-19:
@AndKe the DIY gear doesn't broadcast droneid/remoteid at all. This has been properly verified with an Aeroscope.

Same goes for the O3 Air Unit actually, when otherwise it's more like a commercial drone (DJI Fly etc).

Regarding sniffing nearby traffic: sure, the HW is certainly capable. But given all the info I've seen from people that deal with reversing the commercial drones, there's no evidence this is actually happening, at least with the publicly available firmwares.

AndKe, 2023-05-19:
@j005u Yes, I did not expect remote-id-compliant stuff, but I thought Vista's were sneaking in the serial number somewhere. 
And thank you very much for sharing the reverse-engineering information.
If the activation is purely anti-theft of new devices - then I am totally fine with it.