fpv-wtf/wtfos #80: INSTALL WTFOS BUTTON GREYED OUT

kiddoi, 2023-05-28:
Hello any reason why my button wtfos is greyed out? I can't install it, my goggles v2 is not on v1.00.0606 
https://prnt.sc/b5UnCEnZD77j

LeoniePhiline, 2023-06-09:
- Don’t use all caps
- Are the goggles rooted? You can’t install wtfos without rooting first.

priyanshuhzb, 2023-08-21:
> The button will be grayed out if either 1) you don't have any adb connection to the goggles (need to be rooted) or 2) when health checks that get executed before the install fail to download or run.
> 
> Are you still experiencing this issue? If so can I please get your trace-id from the bottom left of the page?

I have this same issue, device is already rooted.
v2.2.1 - Trace ID: 38f754a3-13e8-45be-ae1f-41bf782a6225

Propjunkie, 2023-10-19:
I am experiencing this same thing. Goggles rooted successfully, twice in fact , both times it's like it connects to the wtf site but the button stays greyed out. Did anyone figure this out????

Wtevo, 2023-10-20:
Same here, please fix....

j005u, 2023-10-20:
It appears we're having server issues since some time late last night. The team is looking into it.

j005u, 2023-10-20:
The issue should be fixed.

Wtevo, 2023-10-21:
Thanks was able to install WTFOS, but now packages don't load in lol

Wtevo, 2023-10-21:
thank you so much

Propjunkie, 2023-10-21:
Hmm ... I ended up using the 06.06 rollback and got it to work that night,
but I would prefer to update to the 1.07 and still be able to use WTFOS.
You're saying that it's fixed now, that should work?

Wtevo, 2023-10-21:
It works, I just did mine 2 hours ago lol

Propjunkie, 2023-10-21:
Sweet. Thanks for the heads up!