fpv-wtf/wtfos #81: I have One Rooted Vista that will connect, but I cannot Update, Or Uninstall

Bwsbmw, 2023-06-03:
i have updated 4 vistas with fairly no issue. i have one vista i have tried to update WTFOS and it will connect to the rooted vista, but will not load the Updates, nor the packages. If i click Packages it tries fetching and fails. its said OPKG is locked. i tried to uninstall wtfos and i cant Uninstall WTFOS. it just feezes at Removing files. The next time i connect they are still rooted. i cannot connect to FPV series as the vista is in a rooted state. I am absoultely lost from here. i have tried multipule comuters. windows 11, 10. no luck. they all connect, but wont let me update, or uninstall wtfos packages. is there a command line promt to remove wtfos? i have no idea what to do, and im not getting very much help on discord unfortunetly.

j005u, 2023-06-13:
Were you able to solve the issue?

If not you should re-flash back to stock with butter and then re-root and re-install wtfos. The latest version should no longer have any problems with the opkg lock. Sorry you had to go through that.

Bwsbmw, 2023-06-13:
Hey Joonas, you can close this. The fix you gave me on discord helped me
instantly. I had been stuck on that one last vista, and it did not want to
play nice. Trying to fetch packages id get OPKG Locked. Thank you for the
quick response in discord with the cli command. It worked right away, I was
way way back on the public release day 0.6.9 package's. All is good,
everything us up to date now Thanks again for your help! -Bwsbmw

closed 2023-06-13 by j005u

cqmiao, 2023-06-18:
me too...
"Failed fetching package list:

OPKG is locked - please wait a bit and reload.

If lock is not released after some time (5 minutes), please reboot your device."

j005u, 2023-06-18:
Try the following command in the CLI: `rm /opt/tmp/opkg.lock` and then restart your device.

This should fix the issue and after updating it should no longer re-occur.

cqmiao, 2023-06-20:
> Try the following command in the CLI: `rm /opt/tmp/opkg.lock` and then restart your device.
> 
> This should fix the issue and after updating it should no longer re-occur.

Thank you bro！