fpv-wtf/wtfos #85: Lost binding after installing wtfos on both goggles v2 and caddix vista

leonard, 2023-06-13:
Hi there,

I've rooted and installed wftos on both: goggkes v2 and caddix vista. Befare installing they were binded normally, but I had a limit with 25mbs which I wanted to expand.

But after install wtfos I have message on my goggles like: image transmittion image lost. I tried to bind it again with no result. How cat I debug this problem?

leonard, 2023-06-13:
After that I tried to remove WTFOS  - the process was ended but the ui was the same, not like it was on original goggles. I tried to flash with butter - interface is just the same. And DJI assistant can't recognize my goggles. I have no ideas what to do to make my setup works again

j005u, 2023-06-13:
Are you absolutely sure you installed V01.00.0606 and not V01.01.0000 on the vista? That would stop pairing with V2 goggles.

How much did the UI change? wtfos install should by default only add custom OSD, the rest of the UI should stay 100% original. Are you sure that you are not in DJI FPV drone mode on the goggles?