fpv-wtf/wtfos #86: Googles V2 freeze after last update

mauroS123, 2023-06-20:
Today after the last update i got 2 freezes on my googles V2, flying different models. The goggles freezes and the only way to get back to video is to unplug and plug again the battery

j005u, 2023-06-20:
@mauroS123 thanks for the report, we take these things very seriously.

How long were these flights? Do you have DVR? Both Goggles side and Air side would be of interest, if you have an original Air Unit.

If you are able to, then I would also like to ask you to see if you can reproduce this behavior on the bench.

mauroS123, 2023-06-20:
It happened today. The flights were over 15-20 minutes long.
One aircraft with AIRUNIT (no SD card no recordings on board) and one aircraft with VISTA.
Both recordings on my googles (video+osd file).
I can check what minute I had to turn on RTH because the video froze.

j005u, 2023-06-20:
Could you perhaps upload the DVR for review? It may offer some insights.

You can e-mail me the links to joonas@batman.ee if you don't want to post them here publicly.

mauroS123, 2023-06-20:
Vista stop at minute 23. Airunit at minute 8

mauroS123, 2023-06-20:
my firmware is 1.00.0607. Do you think I need to downgrade to 1.00.0606  ?

benlumley, 2023-06-20:
>  the only way to get back to video is to unplug and plug again the battery

The goggles battery? Or the airside battery ?

mauroS123, 2023-06-20:
googles battery

j005u, 2023-06-22:
Did the goggles menus also become unresponsive or was it just the video feed that froze?

mauroS123, 2023-06-22:
I dont kmow because i have unplug power

mauroS123, 2023-06-22:
Today, after reinstall wftos and deleting two unusefull package , I haven't had any issue on two long flight

Daniel270667, 2023-07-01:
mauroS123, what "unusefull packages" did you deleted ?

I have exactly the same issue with my Goggles V2 and Caddx Vista both with WTFOS (0.2.4) installed. Also recording on SD-Card. Bandwith was 50 mbps. Lost a quad (P30 with Caddx Peanut) last wednesday because of that. :-(

mauroS123, 2023-07-01:
Update to the latest 0.2.5 and install only 1200mW unlock, auto record, and MPS-OSD.

Daniel270667, 2023-07-01:
Ok, thanks !

j005u, 2023-07-02:
"tweak-fix-bind-loss" and "tweak-prevent-force-upgrade" should not effect this behavior.

wtfos-system 0.2.5 fixed the freezing issue by restoring stock behavior where when the rtos crashes the entire system gets restarted.

We don't know yet if the crash is made more likely by wtfos being installed or not and are investigating the issue. I've only managed 2 crashes with 30+ hours of recording tests so far, so a few test flights here or there are not a good way to validate this isssue.

closed 2023-07-02 by j005u