fpv-wtf/wtfos #87: Wtfos live audio

hondacr1989, 2023-06-30:
Hi :)
Hoping someone can help.
I have install the live audio onto my goggles and the original air unit but I'm not getting any audio back through the headphones.
I have done a playback from the sd card that was in the air unit and its recorded the sound but I can barely hear it?
Thanks

Dcvelopement, 2023-07-07:
Hey,
After I installed the"[djifpv_live_audio_V2](https://github.com/funneld/djifpv_live_audio_V2)" by Funneld I had to turn the volume up. Also make sure to use headphones or earbuds with TRRS connector (like with mic). 
His page explains;
Audio volume can be set:
  1)by playing back a video that doesn't contain any audio and set the volume from there.
  2)with CLI command "setprop persist.dji.volume.headphone 6" with persistent values from 1-6.
  3)with CLI command "tinymix 2 48 48" Values up to 63.

Hope this helps

hondacr89, 2023-07-08:
> Hey, After I installed the"[djifpv_live_audio_V2](https://github.com/funneld/djifpv_live_audio_V2)" by Funneld I had to turn the volume up. Also make sure to use headphones or earbuds with TRRS [...]

Thank you!
I have been using a headset with microphone tried 2x sets and neither worked.
I'll give playing a video back with no audio to adjust volume and try that.
Thank you.

willau01, 2023-11-07:
> Thank you! I have been using a headset with microphone tried 2x sets and neither worked. I'll give playing a video back with no audio to adjust volume and try that. Thank you.

did you get it to work ??