fpv-wtf/wtfos #89: Gogles V2 freezes and restart

mpmdembi, 2023-07-09:
With the newest FW in midflight I noticed one hangs up and restart of V2 gogles. Distance to drone ~20m, 1200mW mod, 50Mbit/sek, Caddx Vista. I had also one freezes without restart on the previous FW.  Before it I dodn't noticed such problem.

Daniel270667, 2023-07-10:
Had the mentioned issue (freezing without restart) with my Goggles V2 and Caddx Vista with WTFOS 0.2.4. Also on 50Mbit/s.
Updated on 0.2.5. Didn't tested yet. After reading this, I consider to totally remove WTFOS from my devices. I already lost a quad because of this issue. I am not sure wether this comes from WTFOS but I don't want to take this risk anymore. And with the same Goggles in DJI FPV drone mode I didn't had any issues so far.

j005u, 2023-07-10:
As per the announcement at https://discord.com/channels/849931046883819550/977165066778587217/1123311486308208831 we don't fully yet understand the root cause of the reboot issue or if wtfos makes it worse - stock can definitely do this under certain conditions.

For the time being uninstalling is the safe bet, although it may still not resolve the issue for you. If so, please do give an update.

We're doing the best we can to chase this down, however due to it being random and difficult to reproduce it's taking longer than anyone would like.

jok250, 2023-08-07:
My experience with this issue is that this only happens after a relatively long flight session without restarting the goggles AND the goggles SD card gets full (wtfos 0.2.5).

It could be the above is inaccurate and I'm just lucky now, but by turning off the goggles when changing the lipo, and keeping always some free space on the SD I have managed to avoid the issue during the last month. Fingers crossed.

Daniel270667, 2023-08-08:
When I experienced this issue with my lost quad, the SD-Card was less than half full. But indeed I didn't shut down the goggles between the flight sessions. I'm now testing without WTFOS and will report about it. I already flew 12 Sessions each about 4 minutes without any issues.

j005u, 2023-08-08:
If anyone is able to further confirm or deny the SD card being related to this then that would be helpful info.

Our own tests have shown that if we write a script to always clear the SD card _before_ it's time to roll over and delete the file then we see less SD card slow errors, but it does not affect the crash rate in 24/7 style recording tests.

Vezrock, 2023-08-09:
Recently, I also encountered a reboot of glasses.  Runcam link wasp nano, Goggles V2 WTFOS 0.2.5.  SD card was less than half full.  I flew 2 batteries each about 10 minutes without any issues.  I didn't shutdown the goggles between the flight sessions.  On 3 batterie, the glasses went into reboot.  It was in an open field and I managed to switch the quadcopter to anglemod.  Generally it was my 2nd FPV flight, I am a beginner and it was not very pleasant.  Now it's scary to lose my only quadcopter.  I have a DVR if that helps.

Lozza0414, 2023-08-11:
My v2 goggles were rebooting nearly every flight even after updating. i rolled back dji firmware on both goggles and air units as well as vistas no change. i read somewhere it may be the sd card so i swapped it managed 6 or more 10min flights before it happened again. not sure where to go now. when it happens dont panick just hit gps rescue then disengage once re booted. hope it gets sorted soon whatever the cause.

destiny1976, 2023-08-13:
My system froze after couple of flights without rebooting the goggles in between.
DVR recording of goggles show after freezing on last frame, the Delay value keeps running, in my situation till >5000ms
so at least not 100% seems to freeze?

btastic, 2023-08-14:
Had this happen today. 700mW. Caddx Vista. Goggles V2. Last frame was slightly above us, I remember that I was able to see a bit more than where the recording stopped. Screen was freezing, then it seemed to reboot. SD card was at ~50%.

Lozza0414, 2023-08-14:
I changed my sd card and I had only 1 reboot from 6 flights. Now I reboot
the goggles after each flight and have had 8 10min flights without a
problem. More testing this morning.

Daniel270667, 2023-08-15:
Yesterday evening I flew 6 batteries à 3,5 minutes without any issue with my V2 and Vista without WTFOS on 50 MBit/s, HD mode.

Lozza0414, 2023-08-15:
It seems wtfos has the bug, rebooting between batteries seems to work for
now.

temtekmedia, 2023-08-15:
I'm still running older version (before the restart bug was announced, I think i have 0.2.1) and have not had issues on the 3-4 months I've been flying this thing. I'm scared to update tho

Lozza0414, 2023-08-15:
Glad to hear it! Don't update till it's resolved.
Cheers.

temtekmedia, 2023-08-15:
To be completely safe would it be good idea to remove sd card or disable auto recording? I just fly close range freestyle so I really dont need the DVR footage

Lozza0414, 2023-08-15:
Yes, you can do that if you're not interested in recording. I have saved my
quads with gps rescue until it reboots then cancelling it. Unnerving when
it happens though.

benlumley, 2023-08-15:
> To be completely safe would it be good idea to remove sd card or disable auto recording? I just fly close range freestyle so I really dont need the DVR footage

our testing showed recording/not recording to make no difference to crash frequency.

benlumley, 2023-08-15:
> I'm still running older version (before the restart bug was announced, I think i have 0.2.1) and have not had issues on the 3-4 months I've been flying this thing. I'm scared to update tho

the bug exists in all versions we've tested; there is currently no known safe version released, old or otherwise. so don't rely on this approach if it's anything critical.

temtekmedia, 2023-08-15:
> the bug exists in all versions we've tested; there is currently no known safe version released, old or otherwise. so don't rely on this approach if it's anything critical.

Good to know! Atleast so far then I've been safe, I might have to knock on a wood now

j005u, 2023-08-16:
Hi all,

We have a new version out for testing that should hopefully solve the issue once and for all.

If you want to join us before a general release you can do so by running the following commands in the fpv.wtf CLI while connected to your goggles:

```
cd /tmp
export http_proxy="http://127.0.0.1:8089"
wget http://github.com/fpv-wtf/msp-osd/releases/download/v0.11.0/msp-osd_0.11.0_pigeon-glasses.ipk
opkg install msp-osd_0.11.0_pigeon-glasses.ipk
reboot
```

Lozza0414, 2023-08-16:
Awesome, thanks for the update!

jok250, 2023-08-17:
Will try this today with a few lipos without rebooting in between, thank you guys!!

devosthomas, 2023-08-18:
@jok250 ... and the verdict is? :)

Lozza0414, 2023-08-18:
I had only 1 15min flight after doing the CLI fix and had no problems. Will
do more testing when the weather clears up.
Cheers.

jok250, 2023-08-25:
> @jok250 ... and the verdict is? 

No more random reboots so far.
I have been flying sporadically this week, and for almost 2 hrs this afternoon (only rebooted the goggles for replacing the battery once).
I am finally on vacations so I should be able to do some more extensive testing during the upcoming days, but this looks promising for now.

btastic, 2023-08-25:
I've also been flying 20-30 packs with the update. No restarts so far. But happened to me only once after like 70 flights in total.

Lozza0414, 2023-08-25:
I also have been testing 8 12 to 18min flights no problems. 3 3 and 2 no
rebooting between flights. Thanks devs.

temtekmedia, 2023-08-25:
official update coming soon or should i run the cli fix on my vtx and goggles? I cant fly now in 4-5 days anyways

Lozza0414, 2023-08-25:
CLI fix has worked for me.

j005u, 2023-08-25:
Thanks everyone for your testing and feedback! We've pushed the update live to the configurator.  Those that are already on the testing version don't need to update, it's the exact same build.

@temtekmedia this update only applies to the goggles side.

Closing, hopefully once and for all!

closed 2023-08-25 by j005u