fpv-wtf/wtfos #95: Feature Request: Auto open AV-IN when goggle power on/ Shortcut in for AV-IN on main menu

dacan2810, 2023-08-28:
Recently I have been using analog for whoop racing. So it is inconvenience when power on goggle and wait until it boots complete then go to the menu and switch to AV-IN.
It would be convenient if we had an option for automatic go to AV-In after boot up. Or at least add a menu for AV-IN outside the Root menu.

j005u, 2023-08-29:
There's an issue open on the wtfos-modloader repo for this here: https://github.com/fpv-wtf/wtfos-modloader/issues/9

The tl;dr; is that it's non-trivial but doable. We're open to PRs implementing, but currently is not a priority on our own todo lists.

Closing duplicate here on this repo.

closed 2023-08-29 by j005u

emulamer, 2023-11-25:
Not sure if this approach is frowned upon, but I made it go to avin by running `setprop dji.factory.avin_test 1` after boot.  I put it in a package in case the quick and dirty solution works for anybody else.  Only tested on the v2 goggles, don't know about anything else. https://github.com/emulamer/avin-startup

dacan2810, 2023-11-28:
Confirm it work on V1 goggle too.
Thanks @emulamer so much. It save me time when manual switch to AV-in when i'm in analog whoop racing.