fpv-wtf/wtfos #103: Goggles 2 Support and maybe bring us the headtracking

ltwin8, 2024-05-28:
Hey there, is there anything in development to support the goggles 2 which have internal IMU and if thats on the roadmap is there any plan to transmit headtracking data to the O3 air unit to enable seamless use of the g-port gimbal or similar?

KTM1551, 2024-09-30:
I don't think they are working on this past the Air Unit/Vista's. But things might have changed I'm not sure. 
Someone part of it or someone that knew of the people behind the major work, said the O3 Air Unit is a lot more involved and i guess with everything on WTF OSD being free, they don't exactly have the free time to do all of this. 

They're not getting paid so im sure with jos, etc there isn't much motivation to work on this.. Plus maybe with something like O3, maybe DJI will step in since it seems like theyre adding a GPS to the O4 Air Unit to make it directly support RID. At least thats what ive heard, i don't think they waste the money on the hardware, but they might not have a choice or might do it to try and not get Banned in the USA, who knows..

Honestly i would gladly pay $20-$30 for a WTF OSD license to see progress made, but i dont think thats happening right now..

benlumley, 2024-09-30:
no plans or work ongoing on anything other than the v1 system (dji au, vista (or link) and v1 / v2 goggles)

closed 2024-09-30 by benlumley